bigtreetech/EBB #115: CAN Klipper reserved pin problem EBB36 / Manta E3EZ

cs301cs301, 2024-04-14:
I have the problem that some of the specified pins are already reserved for CAN. This means that I cannot use the ADXL cs_pin: EBBCan:PB12. Klipper gives error, is reserved for CAN. The E3EZ uses PB12 and PB13 for CAN communication.
I also cannot use the heater_pin: EBBCan:PB13, also reserved for CAN.
So can I not use CAN in this board combination?

Monra124, 2024-04-27:
I have the same problem, I have the ebb with usb communication plus the two power cables, when I turn it on it tells me that PIN PB13 is reserved for SPI2, have you found a solution?.

Thanks

cs301cs301, 2024-04-29:
> I have the same problem, I have the ebb with usb communication plus the two power cables, when I turn it on it tells me that PIN PB13 is reserved for SPI2, have you found a solution?.
> 
> Thanks

I have now switched and use the CAN lines for USB. This works for the E3EZ + EBB36 combination. As for your problem, maybe a pin is named wrong? PB13 is actually the Heater pin on v1.2, SPI2 is PB12. Which boards are you using?

Monra124, 2024-04-30:
> I have now switched and use the CAN lines for USB. This works for the E3EZ + EBB36 combination. As for your problem, maybe a pin is named wrong? PB13 is actually the Heater pin on v1.2, SPI2 is [...]

Exactly the same board, the E3EZ blanket.
With the fimrware installed from github.

Thanks for answering, I've also sent a ticket to BTT, let's see if they answer me

cs301cs301, 2024-04-30:
Then it can only be a config problem... 
my configs:
[config-20240430-173109.zip](https://github.com/bigtreetech/EBB/files/15167182/config-20240430-173109.zip)
my menuconfigs:
E3EZ
[image]
EBB36
[image]

Firmware is Armbian 24.2.3 Bookworm with Linux 6.1.79-legacy-sun50iw9-btt

Monra124, 2024-04-30:
> Then it can only be a config problem... my configs: [config-20240430-173109.zip](https://github.com/bigtreetech/EBB/files/15167182/config-20240430-173109.zip) my menuconfigs: E3EZ ![E3EZ_menuconfig](https://private-user-images.githubusercontent.com/37987673/326878454-eb57118f-836b-4d9b-9dde-22fc7523d07f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ0OTQwNzgsIm5iZiI6MTcxNDQ5Mzc3OCwicGF0aCI6Ii8zNzk4NzY3My8zMjY4Nzg0NTQtZWI1NzExOGYtODM2Yi00ZDliLTlkZGUtMjJmYzc1MjNkMDdmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDMwVDE2MTYxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIwZjc0ODY2ODk3YWJkMjAwMzUyMDNhZmEzYTI2NzYwYjdlNTQyMGI2MTQwNTRlZWEyZTY5NDA1NzQ4YzJhYjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.9qyXBDmpMjBoR9w7dUUZkSJkqlnne7LYUFpnSaXhNPg) EBB36 ![EBB36_menuconfig](https://private-user-images.githubusercontent.com/37987673/326878465-7cf9a6a5-43e9-4222-ace0-1c9b2b4a43f1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ0OTQwNzgsIm5iZiI6MTcxNDQ5Mzc3OCwicGF0aCI6Ii8zNzk4NzY3My8zMjY4Nzg0NjUtN2NmOWE2YTUtNDNlOS00MjIyLWFjZTAtMWM5YjJiNGE0M2YxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDMwVDE2MTYxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI1NzBjNzliMDIzMzdlMmY0N2VhYTUzYjYxZTQzNjUwNzk4NmY5MmZlOTk4YjkzYTEzN2Q2ZThkMmE4ZTdmM2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.nWdqfeNvJUEpWFhrrYaTTPF2JAtSsclEC2T2BUKjSU0)
> 
> Firmware is Armbian 24.2.3 Bookworm with Linux 6.1.79-legacy-sun50iw9-btt

Thanks.
I have done this exactly the same, but when I enter the ebb input I get the PIN error PB13 reserved SPI2

Monra124, 2024-04-30:
I put an older version of the EBB CAN frimware, the error did not come out, I asked to update, it updates everything but the EBB, I downloaded the latest firmware_usb.bin version from the official repository and nothing.

What version of the EBB mcu do you have??

I currently have:
Your Klipper version is: v0.12.0-177-gaf149b47-dirty
MCU(s) which should be updated:
mcu: Current version v0.12.0-177-gaf149b47
EBBCan: Current version v0.12.0-9-gbb4711c5d

Monra124, 2024-04-30:
I updated the firm from menuconfig and the same error comes up again:
[image]

With the configuration:
[https://private-user-images.githubusercontent.com/37987673/326878465-7cf9a6a5-43e9-4222-ace0-1c9b2b4a43f1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ0OTQwNzgsIm5iZiI6MTcxNDQ5Mzc3OCwicGF0aCI6Ii8zNzk4NzY3My8zMjY4Nzg0NjUtN2NmOWE2YTUtNDNlOS00MjIyLWFjZTAtMWM5YjJiNGE0M2YxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDMwVDE2MTYxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI1NzBjNzliMDIzMzdlMmY0N2VhYTUzYjYxZTQzNjUwNzk4NmY5MmZlOTk4YjkzYTEzN2Q2ZThkMmE4ZTdmM2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.nWdqfeNvJUEpWFhrrYaTTPF2JAtSsclEC2T2BUKjSU0](url)

cs301cs301, 2024-04-30:
It seems that you still have the old "fire-hazardous" v1.0 of EBB36. PB13 is there for SPI2.

[image]
For the heater you have to use PB1 in the config.

This is no longer the case with the new v1.2.

[image]

Monra124, 2024-04-30:
The ebb 36 can is v1.2.

The firm I install is the one they have on github v1.1 since it is the same as v1.2.

You won't have the firm you installed on yours to see if that's the problem.

Thank you very much for your time, I really appreciate it as I can't get it to work.

[image]

cs301cs301, 2024-04-30:
That's right, I did not use the pre-built firmware. I followed the instructions for Canboot (now Katapult).

https://wiki.kb-3d.com/en/home/btt/voron/BTT_EBB36

Monra124, 2024-04-30:
Ok, so you have it connected by CAN, instead of USB?

cs301cs301, 2024-04-30:
No, as I said, I use USB. (https://lab4450.com/latest-projects/upgrade-can-ebb36-to-usb-toolhead/)
Initially yes, but that didn't work. I then flashed Klipper over it with USB settings. I just don't know whether any of the old firmware remains on it, because sometimes you specify a bootloader offset, sometimes not. 
I don't really know why it doesn't work for you.
Have you also rebuilt and flashed Klipper for the EBB36 or is it still running the firmware from github? The E3EZ board sees the EBB, so you can simply flash Klipper in DFU moder there and don't need the finished firmware.

Monra124, 2024-04-30:
The board sees the EBB36.
I've tried building the firmware, I've even tried installing RATOS from its installer and the same thing always ends up happening to me, it's as if you always give me a v1.0 signature.

I will take a look at the information you have given me, thank you very much

Monra124, 2024-05-01:
Good morning.
There is no way, no matter what I do the same result, I built the firmware and flashed it in dfu mode, it did it correctly, the result is the same, it's like you always build with the v1.0 board pins.

cs301cs301, 2024-05-01:
Can you pls upload your Klipper configs for EBB & E3EZ?

Monra124, 2024-05-01:
> Can you pls upload your Klipper configs for EBB & E3EZ?

I don't have everything done, as I first tried to link the E3EZ board with the EBB and then connect everything and finish the printer with its macros...

cs301cs301, 2024-05-01:
That's crazy, there is no double reference to PB13, so where is this error message coming from? 
Just out of interest, have you tried entering PB1 for the heater instead of PB13? Not that it would work, but just to see if other error messages appear.

cs301cs301, 2024-05-01:
However, it appears that the board is faulty. Do you have the option of returning it?

Monra124, 2024-05-02:
I opened a ticket on the official BTT website where I bought it, but they don't answer

nexus665, 2024-08-28:
Hi,
I have the same problem - though I am using the EBB36 v1.2 with CAN (self-configured katapult bootloader, not precompiled binary).

As soon as I try to setup the ADXL345, it tells me that PB13 is reserved for spi2 and it cannot start. As above, I have only one reference to PB13, as in:

`[heater extruder]heater_pin: EBB36:PB13
`

`[adxl345] 
cs_pin: EBB36:PB12
`

Only when I comment out the adxl section will Klipper start and connect. 

Otherwise, it gives the exact same error message of:
`Pin PB13 is reserved for spi2`

Monra124, 2024-08-28:
Good morning, my problem was a faulty board, with its replacement it was easier to solve.

duder56, 2024-09-13:
> Good morning, my problem was a faulty board, with its replacement it was easier to solve.

when you say "faulty board" do you mean your main board, u2c, ebb36? 

I have a Manta m8p v1.0, u2c v2.1, and ebb2240 and I'm getting the same error. I don't really see how a new board will fix this issue, as it would be in the coding?. The heater pin will always be PB13 for the heater, as this is how it appears in their schematics. I've tried starting from scratch twice now, it only worked the first install for about 10 min before the first reboot, then all I have been getting afterwards is the PB13 reserved problem. Quite frustrating that using all BTT products, would leave me without a working system...

Monra124, 2024-09-13:
If it was the defective EBB36, I don't know why.

I couldn't tell you, just check that the firmware you introduce is the right one.

duder56, 2024-09-13:
> If it was the defective EBB36, I don't know why.
> 
> I couldn't tell you, just check that the firmware you introduce is the right one.

Ok thanks, at least that is the cheaper board to replace 😅

I've definitely made sure to have all the correct firmware, as it did work temporarily the first install. Then nothing was touched, aside from a reboot, and will not work. I've even gone as far as trying older firmware with fresh installs, and different issues arose with no communication to EBB bus, or the same pin error.

But I will try a new EBB board, thank you for the advice.

nexus665, 2024-09-13:
Hi duder,

did you try the solution I posted a couple of messages above, enabling Software SPI instead of Hardware SPI by defining the 

spi_software_sclk_pin: EBBCan:PB10
spi_software_mosi_pin: EBBCan:PB11
spi_software_miso_pin: EBBCan:PB2

not just 

cs_pin: EBBCan:PB12

on its own? Worked for me...

Of course, you could have a defective board, but hey, worth a try before you buy a new one...

Regards

Monra124, 2024-09-13:
> Hi duder,
>
> did you try the solution I posted a couple of messages above, enabling Software SPI instead of Hardware SPI by defining the
>
> spi_software_sclk_pin: EBBCan:PB10 spi_software_mosi_pin: [...]

I'm sorry I can't help you more.
I hope you fix it

duder56, 2024-09-13:
> Hi duder,
>
> did you try the solution I posted a couple of messages above, enabling Software SPI instead of Hardware SPI by defining the
>
> spi_software_sclk_pin: EBBCan:PB10 spi_software_mosi_pin: [...]

I have a EBB SB2240, and in the sample CFG this is how it is written already. But thank you for the suggestion, obviously I would also like to try all avenues before buying a new board 🤭

duder56, 2024-12-02:
Turns out my issue was stupidity. I had commented out the SPI's under the 2204 driver header, thinking I wasn't running SPI mode, because my 2209 drivers are not in SPI mode 🤦🏻 uncommenting these got ride of my SPI error.