bigtreetech/EBB #124: EBBCan error

wb4608328, 2024-05-28:
MCU 'EBBCan' shutdown: Missed scheduling of next digital out event
This is generally indicative of an intermittent
communication failure between micro-controller and host.Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

How to solve this

Pirat-1989, 2024-06-13:
I faced the same problem on the first layer of a complex part. I changed the firmware to 250,000 and 500,000

orge70, 2024-07-06:
I have the same problem, but no solution

Townie-au, 2024-07-24:
Anyone had any success with this issue. Also experience same issue.

orge70, 2024-07-24:
I changed my pi to rpi4, i read somewhere roi3 should be 32 bit or changed to rpi4, now i have no problem

Townie-au, 2024-07-24:
> I changed my pi to rpi4, i read somewhere roi3 should be 32 bit or changed to rpi4, now i have no problem

Do you mean you changed from a cb1 to a rpi4?
What is rpi4?

Townie-au, 2024-07-24:
I tried changing the rt sample rate as specified in marks wiki but still no joy. 
[klippy.log (16).txt](https://github.com/user-attachments/files/16368391/klippy.log.16.txt)

laborchet, 2024-08-23:
> Do you mean you changed from a cb1 to a rpi4?
> What is roi3?

I think they meant to type "rpi3" 

i read the same forum thread he did, several people running rpi3 switched to the 32bit OS and solved their issue, or run a rpi4 and 64bit OS. 

I am having this same issue, running a rpi3 with BTT kraken and EBB36, no u2c since the kraken has built in canbus, but im having this issue and lots of missed bytes in klippy log. going to try to switch to 32bit OS.