bigtreetech/EBB #127: EBB36 v1.2 MAX31865 RTD input is disconnected

grigi, 2024-06-19:
Hi, Trying to configure my EBB36 v1.2 to use the MAX31865 for a 2-wire PT1000 sensor I get the following errors in my `klippy.log`:
```
Max31865 RTD input is disconnected
Max31865 RTD input is shorted
Max31865 VREF- is greater than 0.85 * VBIAS, FORCE- open
Max31865 VREF- is less than 0.85 * VBIAS, FORCE- open
Max31865 VRTD- is less than 0.85 * VBIAS, FORCE- open
Max31865 Overvoltage or undervoltage fault
```
Doing some research I found that this tends to indicate a wiring fault.

I manually tested the resistance and it's working fine, quadruple-checked that the dip switches are correct and that it's connected to the correct pins, but nothing would fix it.

Then I took a close look at the board, and saw this:
[image]

That ain't right?

grigi, 2024-06-19:
I re-soldered those components (They are tiny) and it resolved the issue.
I suspect they are too close to the mount point and was put under tension.

I modified my mount to leave more clearance at the bottom there so as to not apply tension on those components again.

A longer term solution would be to move those components further away from the screw hole, as this seems to be a relatively common issue.

closed 2024-06-19 by grigi

looxonline, 2024-06-19:
Yea, looking at how they are both angled in a mirrored fashion suggests that whatever screw you put in there somehow pushed them out. Careful with that because it can rip the pads off the board. Glad that you got it sorted out though.