bigtreetech/EBB #131: EBB36 v1.0 wont connect over usb after firmware update.

chimpera, 2024-07-08:
The EBB can be put in dfu mode and the file upload works. It does not show up with lsusb after a reset. This is with both the pre built and compiled firmwares.

jorgejglopez, 2024-09-07:
I'm having the same issue but with de EBB42. The pre built firmware its detected but didn't work with Klipper because it needs to be updated. The compile firmware with the makemenu didn't work. It's possible to upload the firmware but didn't show with lsusb. Any solution?

jorgejglopez, 2024-09-07:
After a lot of trial and error I have a solution for my problem. With USB connection I need to compile de firmware with these options:

Enable extra low-level configuration options: check
Micro-controller Architecture: STMicroelectronics STM32
Processor model: STM32G0B1
Bootloader offset: No bootloader
Clock Reference: 8 MHz crystal
Communication interface: USB (on PA11/PA12)

Mi error was The Bootloader offset, I was trying with 8KiB Bootloader