bigtreetech/EBB #135: Sb2209 usb v1 no boot mode

aee74, 2024-09-04:
I’ve tried to hold down boot press reset and let go of boot but no change when I loom at lsusb

how to get into boot mode. Is the board defective?

Aibehn, 2024-09-10:
I'm having the same problem.
- I supplied 24v to the adapter board, connected USB-C to RPi Host.
- Connected XT30(2+2) to both ends.
- Power led indicator goes ON. and a big STATUS led goes on RED.
- I hold press BOOT bottom, press & release RST buttom and after that I release the BOOT bottom.
STATUS led goes  off. POWER led still fine.
It should be on DFU now, but I don't see it with lsusb, or dfu-util -l

looxonline, 2024-09-10:
Could you both list your lsusb results before and after the process?

Aibehn, 2024-09-10:
> Could you both list your lsusb results before and after the process?

`Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 1d50:614e OpenMoko, Inc. stm32h743xx
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

No changes.

looxonline, 2024-09-10:
> > Could you both list your lsusb results before and after the process?
> 
> `Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 001 Device 007: ID 1d50:614e OpenMoko, Inc. stm32h743xx Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
> 
> I have a SKR3 EZ connected. I will try disconnecting everthing. This produced no changes. Just removed stm32 line.
> 
> I have to mention that I checked continuity between DP & DM pins on the toolhead board all the way to the USB-A connector.

That lsusb is before or after going into boot mode or is there zero difference between the two?

Aibehn, 2024-09-10:
``

> That lsusb is before or after going into boot mode or is there zero difference between the two?

ZERO difference

looxonline, 2024-09-10:
Also what USB-C cable are you using between the SBC and the adapter PCB? Is it the one supplied with the part or your own? Have you tried using other cables?

Aibehn, 2024-09-10:
> Also what USB-C cable are you using between the SBC and the adapter PCB? Is it the one supplied with the part or your own? Have you tried using other cables?

I'm using the one supplied. But I did also test a different usb-c to usb-a cable. Same results.

looxonline, 2024-09-10:
OK, fire off an email to luke@bigtree-tech.com and I'll ask the support team to look deeper into your case.

Aibehn, 2024-09-11:
Just a quick update because things escalated quickly.

I was testing it with another host, an RPi2B that I had lying around. At first, no changes, just the same result: no way to get it into DFU mode or recognized by the RPi.

At first, it seemed that the RPi was having power problems. I noticed that plugging in the USB-C cable would cause the power status LED on the RPi to flicker. But then it got worse.

I tried another USB-C cable, and at first, nothing seemed to happen—everything was fine, no magic smoke. After that, it started to smell, and I noticed that the SB2209 was getting hot. Still, no lsusb device, and the status LED was fine.

I tried one more time with the boot/rst button combo, and the LEDs just went off—no magic smoke, nothing. Then I noticed the RPi power LED going crazy. The SMSC chip controller (USB hub & LAN) was getting extremely hot. I disconnected everything, but it was too late.

The RPi2B doesn't boot anymore. The USB hub controller chip gets extremely hot.

The SB2209 doesn't show any LED status anymore. It gets 24V through the adapter board, but nothing happens. No heated components either.

At least It was not the RPi4 and a RPi2B that died. I not hooking anything without a USB isolator or something similar.

tl;dr: SB2209 died and It killed a RPi2B on the process.

looxonline, 2024-09-11:
> Just a quick update because things escalated quickly.
>
> I was testing it with another host, an RPi2B that I had lying around. At first, no changes, just the same result: no way to get it into DFU [...]

Please take the exact connection that you have, remove the EBB USB from one end and the Pi from the other and measure to see if the data pins are shorting to any of the power pins or each other.

anilfrace, 2025-01-07:
Has anybody solved this issue?
I am facing exactly the same issue, can somebody guideline me to solve it?

jace, 2025-01-27:
I have a new SB2209 USB and it refused to go into bootloader mode, same as the other reports here. After a couple attempts at boot+reset however, it showed up in `lsusb` and in `/dev/serial/by-id`:

```
Bus 001 Device 016: ID 1d50:614e OpenMoko, Inc. rp2040
```
I then tried to flash it directly using `make flash FLASH_DEVICE=/dev/serial/by-id/usb-Klipper_rp2040_xxxxxxx` but that failed:

```
  Flashing out/klipper.uf2 to /dev/serial/by-id/usb-Klipper_rp2040_xxxxx
Entering bootloader on /dev/serial/by-id/usb-Klipper_rp2040_xxxxx
Device reconnect on /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.4/busnum
Traceback (most recent call last):
  File "/home/pi/klipper/./scripts/flash_usb.py", line 390, in <module>
    main()
  File "/home/pi/klipper/./scripts/flash_usb.py", line 387, in main
    flash_func(options, args[0])
  File "/home/pi/klipper/./scripts/flash_usb.py", line 340, in flash_rp2040
    flash_picoboot(options.device, binfile, options.sudo)
  File "/home/pi/klipper/./scripts/flash_usb.py", line 202, in flash_picoboot
    with open(usbdir + "/busnum") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.4/busnum'
make: *** [src/rp2040/Makefile:78: flash] Error 1
```

Instead, it _now_ showed up in bootloader mode in `lsusb` and I was then able to flash it with `make flash FLASH_DEVICE=2e8a:0003`. Now it works.

jace, 2025-01-27:
I gave my board a test-run by turning the hotend up to 150°C. It worked for about a minute and then shutdown. The board no longer shows up in `lsusb`, even after a few power cycles. I changed the USB cable to a known-good cable but that didn't help either. The LED on the board no longer changes color/intensity when I press boot+reset.

Looks like I have a dead board too?

jace, 2025-02-01:
Turns out I have a bad cable with a loose contact in one of the plugs, so I sliced it open to inspect:

[image]

I'm not an electronics professional, but is that solder on crimped wires? Doesn't that make it brittle and prone to damage like this? And is that shielding sleeve on the middle neutral wire electrically conductive? It's got strands sticking out dangerously close to the 24V wire.

looxonline, 2025-02-01:
> Turns out I have a bad cable with a loose contact in one of the plugs, so I sliced it open to inspect:
>
> [image]
>
> I'm not an electronics professional, but is that solder on crimped wires? Doesn't [...]

There is no crimping. The XT30 connectors are entirely soldered. The internals are potted with glue to prevent the shielding from touching surrounding wires.

jace, 2025-02-04:
You're right. I got a spare XT30 2+2 plug and that takes crimped leads, but the SB2209 USB cable is just soldered. The spare has lock holes for crimped leads.

I've resoldered and hot glued the leads on mine and it's now working. Hope this holds until there's a replacement.

[image] [image]