bigtreetech/EBB #136: SB2209 USB need modded STLs

Steiner-SE, 2024-09-10:
As is the SB 2209 board is not even remotely compatible with the Galileo 2 extruder (and probably any related extruder) as the back USB port cuts into the main body. I found a third party modded main body that partly solves the issue, but the board end up installed angles and under some stress as it catches on the top part and end up installed at an angle (potentially misaligning the connectors for the sub board). Likely a new main body and door is needed, and with the higher XT30 connector a new umbilical solution.
[image]

looxonline, 2024-09-10:
Hi there. Currently the EBB USB is only compatible with the CW2 extruder. Nevertheless, we have one of our structural engineers working on Galileo2 files and will upload them to github as soon as they are done. May still be a week or so.

looxonline, 2024-09-11:
> As is the SB 2209 board is not even remotely compatible with the Galileo 2 extruder (and probably any related extruder) as the back USB port cuts into the main body. I found a third party modded main body that partly solves the issue, but the board end up installed angles and under some stress as it catches on the top part and end up installed at an angle (potentially misaligning the connectors for the sub board). Likely a new main body and door is needed, and with the higher XT30 connector a new umbilical solution. ![2024-09-10 08 11 20](https://private-user-images.githubusercontent.com/132791331/365938757-b9df8452-c8ad-4d71-9b0b-d892673e37bb.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYwNjkyMjUsIm5iZiI6MTcyNjA2ODkyNSwicGF0aCI6Ii8xMzI3OTEzMzEvMzY1OTM4NzU3LWI5ZGY4NDUyLWM4YWQtNGQ3MS05YjBiLWQ4OTI2NzNlMzdiYi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxMVQxNTM1MjVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ZGM5NWYzYjRmMTY0Y2ExMWFkMjM1NzU0YTc0YmFjYjg4ZWI3YTkyZWIxZGE4ZTQxOWYwOGQyMjY2NzY1ZjM1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-ZWUCzIdkpC_ASaq_AvGCdm_UkIJUgpcnaRbhluFD-4)

I have an untested set of files from the designer that will take us about 1 week to test and possibly refine. If you would like to beta test them then let me know.

spooknik, 2024-09-28:
> I have an untested set of files from the designer that will take us about 1 week to test and possibly refine. If you would like to beta test them then let me know.

Hello! I just recieved the SB2209 USB and I will soon recieve the Galileo 2 parts. I wouldn't mind beta testing the parts.