bigtreetech/EBB #137: rp2040 will not go into DFU Mode

Charleslotto, 2024-09-11:
I have all connections disconnected except USB. did the hold boot and unplug/plug in the usb cable and still nothing. 

I see below and get an error:
biqu@BTT-CB1:~/klipper$ lsusb
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 004: ID 1d50:614e OpenMoko, Inc. stm32g0b1xx
Bus 002 Device 015: ID 2e8a:0003 Raspberry Pi RP2 Boot
Bus 002 Device 003: ID 04d9:8030 Holtek Semiconductor, Inc. BTT-HDMI5
Bus 002 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
biqu@BTT-CB1:~/klipper$ make flash FLASH_DEVICE=2e8a:0003
Flashing out/klipper.bin to 2e8a:0003
sudo dfu-util -d ,2e8a:0003 -R -a 0 -s 0x8000000:leave -D out/klipper.bin

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

Failed to flash to 2e8a:0003: Error running dfu-util

If the device is already in bootloader mode it can be flashed with the
following command:
make flash FLASH_DEVICE=0483:df11
OR
make flash FLASH_DEVICE=1209:beba

If attempting to flash via 3.3V serial, then use:
make serialflash FLASH_DEVICE=2e8a:0003

make: *** [src/stm32/Makefile:111: flash] Error 255

looxonline, 2024-09-11:
dfu-util is a tool for STM microcontrollers and will not work with RP2040 microcontrollers. Please follow the respective product manual for different programming options based on the MCU type.

closed 2024-09-11 by looxonline

Charleslotto, 2024-09-11:
I guess I didn't describe it right, I am trying to do the Canbus SB2240 and SB2209 setup but get to the flashing of SB2240 when I put it in what should be DFU mode it shows up as 2e8a:0003 Raspberry Pi RP2 Boot or 1d50:614e OpenMoko, Inc. rp2040 when I just power it on.

looxonline, 2024-09-11:
The SB2209 and SB2240 (non RP2040 variants) are using STM chips. The address you therefore want to use with DFU util is: 1d50:614e OpenMoko, Inc. stm32g0b1xx