bigtreetech/EBB #138: extruder issues

Machine73, 2024-09-13:
I am using the new EBB SBB 2209 USB tool board.
For the life of me I cannot get my extruder to extrude properly since swapping to the new board. Already having issues with the beta z-offset, but I can't solve that issue until I solve the issue of under extrusion. I can manually extrude fine, I checked my rotation distance and it's damn near on point, I swapped to my original extruder from my stealthburner, and verified that nothing is binding with the gears and rollers. But when I go print the first layer is super thin (due to improper z-offset) but as soon as it hits the second layer and hits print speed, it prints spider threads.

bigtreetech, 2024-09-13:
May be internal clogging due to thin first layer. Either that or a config issue. Not hardware related so will leave open for a few more hours for others to comment and then close.

Machine73, 2024-09-16:
SOLVED:
I solved the issue by moving the extruder values from the sample config directly into my printer config. 
Was able to repeat the issue by going back and forth regarding extruder values location. 
Machine runs properly now

fargkenob61-lang, 2025-08-17:
Solved, changed Stealth Chop back to 0.