bigtreetech/EBB #146: SB0000 fans not working

SunayKichukov, 2024-12-24:
I have seen this somewhere with other users having the same issue, the fans connected to the SB0000 just don't run
can somebody address this issue because I spend 2 days taring to find the error but it seams to be the either of the SB000 board or the EBB SB2209.

bigtreetech, 2024-12-24:
Please make sure that you have a proper connection between the main board and the SB0000. Often users have a pin offset or the pins are not reaching deep enough into the main board. Check your pin offset and then also apply pressure to the front of the stealth burner with one of the fans turned on to see if that starts it spinning. Over tightening the bottom screws on the stealth burner can cause this bad connection.

SunayKichukov, 2024-12-24:
I have tested the connections and I ordered another SB0000 just to see if the board is the problem. Unfortunately they both behave the same. I have Heater, extruder motor, BLtouch and a pt1000 sensor hooked to the SB2209 and they all work. The part cooling fan runs at max voltage at whatever power the jumper is set at and the heatsink fan doesn't run at all. I checked the wiring and the pinout only 20 times so I believe that is not the problem, The pinout is set as follows:

[fan]
pin: EBBCan: PA1

[heater_fan hotend_fan]
pin: EBBCan: PA0
max_power: 1.0
fan_speed: 1
kick_start_time: 0.1
heater: extruder
heater_temp: 30.0

The main board is octopus MAX EZ
running on BTT pi v2

bigtreetech, 2024-12-24:
The SB2209 simply outputs PWM signals directly from the microcontroller to the MOSFETS on the SB0000. If the SB0000 was inserted with incorrect pin alignment or inserted/removed with the power to the machine on at any point then it is possible to cause damage to these pins on the MCU and that may result in fans either staying on permanently or not turning on at all. It is best to contact our support team using the ticketing system on the website as github is not one of our official support channels. It is mostly used as a means for engineers to interact with customers when gathering product feedback.

closed 2024-12-24 by bigtreetech