bigtreetech/EBB #147: EBB SB2209/2240

drjeh, 2024-12-27:
CanBus seems to be working except for the hotend fan. Mainsail says the fan is activated, but registers 0 rpm, and the fan isn't running. Parts fan works fine. 

My config for the fan looks like this:
[fan]
pin: EBBCan: PA1

[heater_fan hotend_fan]
pin: EBBCan: PA0
heater: extruder
heater_temp: 50.0

#[heater_fan 4W_FAN0]
pin: EBBCan: PB14
tachometer_pin: EBBCan: PB15
tachometer_ppr: 1

Any ideas?

Thanks

Joel

igeighty, 2025-01-06:
test that the fan actually hasnt left the chat., mine died i chased the same symptoms for a while before realising the fan was toast

ThorSc, 2025-01-08:
Hello,
I have the same issue. Manta M8P - SB2209/SB0000 - CAN bus.
Tested both fans on EBBCan:PA1 and they are working on this pin. Only on EBBCan:PA0 the fans are not working.
Any ideas?