bigtreetech/EBB #149: SB2209 RP2040 MCU 'EBBCan' shutdown: Thermocouple reader fault

mylastregret, 2025-01-16:
[extruder]
step_pin: EBBCan:gpio18
dir_pin: !EBBCan:gpio19
enable_pin: !EBBCan:gpio17
##  Update value below when you perform extruder calibration
##  If you ask for 100mm of filament, but in reality it is 98mm:
##  rotation_distance = <previous_rotation_distance> * <actual_extrude_distance> / 100
##  22.6789511 is a good starting point
rotation_distance: 22.6789511   #Bondtech 5mm Drive Gears
##  Update Gear Ratio depending on your Extruder Type Use 50:10 for Stealthburner/Clockwork 2
##  Use 50:17 for Afterburner/Clockwork (BMG Gear Ratio)
##  Use 80:20 for M4, M3.1
gear_ratio: 50:10
microsteps: 16
full_steps_per_rotation: 200	#200 for 1.8 degree, 400 for 0.9 degree
nozzle_diameter: 0.400
filament_diameter: 1.75
heater_pin: EBBCan:gpio7
##	Validate the following thermistor type to make sure it is correct

## Check what thermistor type you have. See https://www.klipper3d.org/Config_Reference.html#common-thermistors for common thermistor types.
## Use "Generic 3950" for NTC 100k 3950 thermistors
sensor_type: MAX31865
sensor_pin: EBBCan:gpio9
spi_software_sclk_pin: EBBCan:gpio10
spi_software_mosi_pin: EBBCan:gpio8
spi_software_miso_pin: EBBCan:gpio11
min_temp: 10
max_temp: 270
max_power: 1.0
min_extrude_temp: 170
rtd_nominal_r: 1000
rtd_reference_r: 4300
rtd_num_of_wires: 2
control = pid
pid_kp = 26.213
pid_ki = 1.304
pid_kd = 131.721
##	Try to keep pressure_advance below 1.0
#pressure_advance: 0.05
##	Default is 0.040, leave stock
#pressure_advance_smooth_time: 0.040

##	E0 on Motor7
##	Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 extruder]
uart_pin: EBBCan:gpio20
interpolate: false
run_current: 0.650
sense_resistor: 0.110
stealthchop_threshold: 999999

#[filament_switch_sensor material_0]
#switch_pin: PC1

#[filament_switch_sensor material_1]
#switch_pin: PC2

##########################

CharlesNoScreen, 2025-02-03:
How did you solve it?

closed 2025-02-18 by bigtreetech

kingbob2, 2025-03-03:
same issue. solution? I'm suspecting bad physical connection.

mylastregret, 2025-03-05:
Are you using a PT1000 or PT100?

kingbob2, 2025-03-05:
PT1000. It was indeed a bad physical connection. Tore my stealthburner hotend down and found I pinched a thermistor wire.