bigtreetech/EBB #151: Main Body and Cable Cover for SB2209 RP2040 Version?

CookiezRGood, 2025-01-23:
I can't find files for the different stealthburner body and cable cover to fit this board in my toolhead. I tried printing the one for the SB2209 since that section has STL files but it doesn't fit. The PDF for assembly shows a different cable cover model but I can't find the file anywhere.

phenolplus, 2025-02-25:
Same issue. [Cable_Cover_EBB.stl](https://github.com/bigtreetech/EBB/blob/master/EBB%20SB2240_2209%20CAN/STL/Cable_Cover_EBB.stl) is very tight on the top. It's pinching the probe wires when one is plugged in.

CookiezRGood, 2025-02-25:
I actually modified my own version in Blender to make it fit and it works great. I just grabbed all the middle points and extended them out a bit while leaving the edges the same so it still lines up with the body they provide. I uploaded it to Thingiverse and have the link below.

https://www.thingiverse.com/thing:6961699

closed 2025-02-25 by CookiezRGood