bigtreetech/EBB #152: SB2209 USB and PT1000 thermistor

jace, 2025-01-27:
I have a PT1000 sensor in my Rapido Hotend 2, which I've connected to the TH0 port of SB2209 USB, with the switch pushed up to `PT` side. My Klipper is configured as per the example config:

```ini
[extruder]
pullup_resistor: 2200 # 2.2K
sensor_type: PT1000
sensor_pin: EBB:gpio26
```

However, Klipper fails to start reporting ADC out of range (21000+°C). If I remove `pullup_resistor`, it goes even higher. What is the correct setting for PT1000?

jace, 2025-01-27:
Resolved by pushing the switch down to the NTC side. Now it reports temperature correctly. However, the board has shutdown entirely after a few minutes and no longer shows up as a USB device (similar reports in #135).

Please fix the documentation on PT1000 vs NTC as the diagrams in the manual and on the board are not clear at all. Is this supposed to mean I have to select NTC even for PT1000? What is the purpose of the switch then?

[image]

jace, 2025-01-27:
I think I understand now:

If the switch is set to PT, it requires the MAX31865 chip and the additional switches (PT100 or PT1000) to be soldered onto the board, as shown in the bottom right corner of the image. Without them, it's just cutting off the signal.

Leaving this comment here for the next person who's similarly confused.

closed 2025-01-27 by jace

Karl-Barx, 2025-02-13:
What was your solution for this? I'm about to ditch the pt1000 on my rapido for an ntc but really don't want to since it will require redoing my faceplate for sb. Any help is greatly appreciated!

Karl-Barx, 2025-02-13:
I'm just very confused since the market the sb2209 as being pt1000, give you almost no information, and then leave you out to dry. Very frustrating

bigtreetech, 2025-02-13:
The manual contains all of the instructions. Set the switch to NTC and then configure the PT1000 within klipper.

[image]

Karl-Barx, 2025-02-13:
You should check you discord. I'm not the only one with this issue. I've heard from multiple people that you must solder a switch and max chip to get it working. Is this true?

bigtreetech, 2025-02-13:
No, it's not true. You just plug it in, follow the manual guidance on what to set the switch to and configure klipper for a PT1000 thermistor with a 2k2 pull up.

wkeifenheim, 2026-01-19:
I also have a PT1000 from a Rapido 2 hotend, with the same issue as OP, however I do not get usable temperature readings even in `NTC` mode.

To echo the OPs question

> Is this supposed to mean I have to select NTC even for PT1000? What is the purpose of the switch then?

When I select The `NTC` setting for the switch, at room temperature (~20C) the PT1000 reads at ~376.125C. When I select the `PT` setting for the switch, the temperature measures 1997714.827C.

My settings are the same as OPs first post:

```yaml
pullup_resistor: 2200 # 2.2K
sensor_type: PT1000
sensor_pin: EBB:gpio26
```

It is possible I have a bad PT1000, however the more I look into the posts and information about this board it seems like the "supports PT1000" statement on the website product page should be removed, or stronger caveats should be added to the [wiki instructions](https://global.bttwiki.com/EBB%20SB2209%20USB.html#100k-ntc-or-pt1000). The latest V1.2 instructions list the two-wire `TH0` port as `TH0-NTC`. Why bother having a dip switch for `NTC<>PT` if not providing instructions to self-source and install the missing MAX31865 chip and PT100/PT1000 dip switch? This product is confusing and not one I would recommend to others with the current state of documentation.