bigtreetech/EBB #155: USB Errors when homing with SB2209 USB

keyz182, 2025-02-20:
When homing Z, as soon as the head reaches the X/Y coords, this error is shown in the output from `dmesg` corresponding to the SB2209 hub

```
[  851.124515] usb 1-3: clear tt 1 (91b1) error -71
[  851.124719] usb 1-3: clear tt 1 (91b1) error -71
[  851.124992] usb 1-3: clear tt 1 (91b1) error -71
[  851.125473] usb 1-3: clear tt 1 (91b1) error -71
[  851.125655] usb 1-3: clear tt 1 (91b1) error -71
[  851.125946] usb 1-3: clear tt 1 (91b1) error -71
[  851.126346] usb 1-3: clear tt 1 (91b1) error -71
[  851.126403] usb 1-3: USB disconnect, device number 19
[  851.126410] usb 1-3.4: USB disconnect, device number 20
[  851.380032] usb 1-3: new full-speed USB device number 23 using xhci_hcd
[  851.495156] usb 1-3: device descriptor read/64, error -71
[  851.712069] usb 1-3: device descriptor read/64, error -71
[  851.935038] usb 1-3: new full-speed USB device number 24 using xhci_hcd
[  852.051918] usb 1-3: device descriptor read/64, error -71
[  852.271289] usb 1-3: device descriptor read/64, error -71
[  852.373998] usb usb1-port3: attempt power cycle
[  852.760990] usb 1-3: new full-speed USB device number 25 using xhci_hcd
[  852.761127] usb 1-3: Device not responding to setup address.
[  852.965096] usb 1-3: Device not responding to setup address.
[  853.172869] usb 1-3: device not accepting address 25, error -71
[  853.172989] usb 1-3: WARN: invalid context state for evaluate context command.
[  853.286857] usb 1-3: new full-speed USB device number 26 using xhci_hcd
[  853.287025] usb 1-3: Device not responding to setup address.
[  853.493078] usb 1-3: Device not responding to setup address.
[  853.700841] usb 1-3: device not accepting address 26, error -71
[  853.700972] usb 1-3: WARN: invalid context state for evaluate context command.
[  853.701108] usb usb1-port3: unable to enumerate USB device
```

And these logs from klippy.log - https://gist.github.com/keyz182/8b07d0463c88b53338cec88a898f6e0a
I've tried it on different ports on the computer with no luck. 

I did have a Beacon3D RevH and a 3DO V2 camera plugged in - for testing I have removed the camera, and wired the Beacon in a separate cable rather than through the SB2209 USB hub to eliminate variables. Still the same result with no USB devices plugged into the SB2209. 

I am using the supplied cable from the SB2209, and have installed the grounding wire.

bigtreetech, 2025-02-20:
It may be cable related. Could you try to wrap a small cable tie around the connector like this and see if it makes a difference? Sometimes the cable moves within the socket and causes microsecond disconnects.

[image]

keyz182, 2025-02-20:
[image]
[image]

This is about the best I could manage, but it still errors out.

keyz182, 2025-02-20:
Even tried a small dab of glue to try to hold it in place better (second photo you can see it). No luck

keyz182, 2025-02-20:
I think the cable might be damaged. I can't get it to connect at all now - just that same error in `dmesg` each time I unplug/plug in, even after taking the cable out of the chain.

keyz182, 2025-02-22:
Confirmed the cable is bad - no continuity on the DP line. Must have been small enough a break that it worked for a while then slowly moved apart until there's no contact at all now. Will get a new cable.

keyz182, 2025-02-22:
Different cable, same error, so I must have just badly meaured the old one or something. An LDO Nitehawk connects and works correctly over this cable if I swap it over. So I guess it must be board related?

andzymonek, 2025-02-28:
https://github.com/bigtreetech/EBB/issues/157

check the quality of the solders, you can see that there is a bigger problem with it

keyz182, 2025-02-28:
@andzymonek I did reflow those joints, to no avail unfortunately. I've contacted support as I suspect something on the board is dead. My guess is the circuitry driving the onboard USB hub, or the usb hub chip itself. Total guess though.

closed 2025-03-20 by keyz182