bigtreetech/EBB #156: Pinheader Type EBB SB2209

Battlecake91, 2025-02-23:
I broke my pinheader from the Main PCB of the SB to the Fan-Breakout. I cannot find any possible Pinheaders. I tried some I have at home, but they don't fit as the original pinheaders have 0.5mm diameter and normal square ones have 0.64mm. 
I tried then one that fits with 0.64mm but it is too long and now nothing fits.

Does anyone now what part is used original?

Johnnytk36, 2025-04-04:
The generic name is Dunpont 2.0mm spacing. Though in my research they could be a knockoff of molex MilliGrid connectors also.

I know because I wanted to make a male to female jumper to do some testing on y fan board related to the tach line feedback. 

[image]

You could get the connector here. [2.0mm Double Row Right Angle Pin Header Strip Male Pin Copper Needle Nickel Plated](https://www.aliexpress.us/item/3256802252624530.html?pdp_npi=4%40dis%21USD%21US%20%240.98%21US%20%240.90%21%21%210.98%210.90%21%402101effb17437822869141217ef170%2112000021028762605%21sh%21US%21138767790%21X&spm=a2g0o.store_pc_allItems_or_groupList.new_all_items_2007522140976.1005002438939282&gatewayAdapt=glo2usa)

Battlecake91, 2025-04-04:
Thanks for your reply, but I'm sure it is 2,54mm. 

Also the problem is, that the linked pinheader doesn't fit because of the length of the pins. I had some and tried to connect it with them, but you cannot put on the cover completelly after doing this.

Johnnytk36, 2025-04-04:
That's weird. Yeah, i don't know which one you have then. As the spec sheets show 14 pins like the one i have.  The making board SB0000 is the same for the SB2209 and SB2240 it shows also. 

[BIGTREETECHEBB SB2209 CAN V1.0_IO.pdf](https://github.com/user-attachments/files/19607779/BIGTREETECHEBB.SB2209.CAN.V1.0_IO.pdf)

[BIGTREETECH EBB SB0000 CAN V1.0_size.pdf](https://github.com/user-attachments/files/19607806/BIGTREETECH.EBB.SB0000.CAN.V1.0_size.pdf)

Johnnytk36, 2025-04-04:
Nevermind found it. I haven't updated my files on my laptop in a while, and i didnt show the have the files for the SB2209 USB version. The USB version has 10 pins. 

[BIGTREETECH EBB SB2209 USB-Pin.pdf](https://github.com/user-attachments/files/19607849/BIGTREETECH.EBB.SB2209.USB-Pin.pdf)