bigtreetech/EBB #157: Sb2209 Bad solder connection

24Tigger24, 2025-02-24:
I have connection issues with one of my Sb2209. Wiggeling the connector i can See a little spark on GND.

Bad Connection:
[image]

(better) Connection:

[image]

andzymonek, 2025-02-28:
did you have communication errors in the klipper or did you just notice sparking

24Tigger24, 2025-02-28:
Both

andzymonek, 2025-02-28:
After fixing those solders did the problems disappear or did you do something else?

24Tigger24, 2025-02-28:
I added a bit of solder (GND was difficult because it didn't warm up) and now it doesn't connect via Canbus to the printer it should go in. But still connects with my other printer. Maybe i somehow killed the U2C...

Hooba97, 2025-02-28:
i faced a similar problem, the inner small connectors for data bent as soon as i plugged in the cable
i got lucky i have a friends who fixed it for me 
i hope this feedback helps for the future