bigtreetech/EBB #162: Pi does not boot when the SB2209 is plugged in

cmeyer231, 2025-03-31:
Pi does not boot when the SB2209 is plugged in, but once I unplug the SB2209, either with the USB C or the XT30 it boots fine. I can plug the USB in right away and nothing seems to be wrong, the board and PI operate 100% normally.

I have tried Kapton tape over the 5V pin on the PI side of the USB cable, different USB cables and the same issue occurs. 

This was a replacement for the previous SB2209 I had that kept getting "timer too close" errors. This replacement seems to have fixed that problem but now the PI booting issue is now.

I though wiring on the hot end side might be bad, but the board causes the issue without anything plugged in at all other than the XT30.

Compared to the random TTC errors causing the print to stop 2-3 hours in this a a more minor frustration, but still would like to get it fixed.

Pi being powered from my Spider 2.2 board and a separate 24 to 5v buck converter (both rated at 3 amps) - same issue.

asyncmeow, 2025-04-13:
saw this issue while checking to see if someone asked another here about something i was wondering and saw this. after having a similar issue with my Octopus Pro board earlier - what model pi are you on? the 4b has an issue where if power is getting backfed from a USB port, it doesn't boot. i was able to resolve this earlier by adding a USB hub in-between the pi and board, but one way you might be able to check is by connecting the pi to a display while it's booting - when you power up the pi, does it reach the firmware of the pi before you unplug the SB2209? or is it stopping later on in the boot process?

for context, here's a video of the issue i had - does your issue look similar?:

https://github.com/user-attachments/assets/ecaee532-677a-40af-8ea5-bd75c6dc929d

cmeyer231, 2025-04-13:
It is exactly the same as what I have. Noting happens on screen until the toolhead is unplugged.

I thought it was power backed as well, so only the data lines are connected. I used kapton tape on the pins, and verified there was no voltage on the USB c side when properly powered.

I can also unplug the board from the toolhead side and it boots, so it definitely seems like something with the data lines, since power would still be backfed in that case. 

A USB hub controller between the two would make sense if there is something the rpi is doing that is breaking it. I may try and use an old version of klipper and see if there is a bug introduced in a recent version.

asyncmeow, 2025-04-13:
the thing that makes me unsure if it's a klipper specific issue is that i have other boards running klipper that i can leave plugged in while it's booting without issue, but something with the data lines could be another explanation for it - just not sure how to debug it... i have a usb packet analyzer somewhere, but no way to run its software anymore (or the cables i need for it) and its usb1.1 only so i am unsure if it would help here 😅

Bersaker, 2025-05-13:
I have exactly the same problem with the skrmini e3v3 and the ebb42. If I connect both usb connections with the rpi4b he does not boot. If I pull a plug, it starts to boot. Does anyone know why the pi4b does this?
I tried the zero w2, rpi3b, rpi3b+ and rpi5. Everything works without any problems.
There must be a solution for the pi4b...