bigtreetech/EBB #164: Lost communication with MCU 'EBBCan' after 1-3 minuts of printing

ZlobniyDaemon, 2025-04-11:
Recently encountered the following problem:  
After 1–3 minutes of printing, the connection with the EBB SB2209 RP2040 is lost and the printer shuts down.  
This issue only occurs during printing. In all other situations (toolhead movements, hotend or heat bed heating, filament extrusion), no errors occur.

SB2209 is connected to Octopus Pro V1.1 via CAN. I also tried connecting through BTT U2C, changing the CAN cable, flashing the controller, but the error persists and only occurs during printing.

[klippy (1).log](https://github.com/user-attachments/files/19706790/klippy.1.log)

bigtreetech, 2025-04-11:
It seems like you may have one of two issues:

1. Poor strain relief on the connector leading to comms failures while moving - Using a cable tie around the connector and the socket to lock it in place can help with this.
2. ESD discharge into the EBB resulting in an MCU reset and a disconnect event. To solve this you can have a look at the EBB SB2209 USB manual where there is a section on grounding.

ZlobniyDaemon, 2025-04-11:
> It seems like you may have one of two issues:
>
> 1. Poor strain relief on the connector leading to comms failures while moving - Using a cable tie around the connector and the socket to lock it in [...]

bigtreetech, 2025-04-11:
> > It seems like you may have one of two issues:
> > 
> > 1. Poor strain relief on the connector leading to comms failures while moving - Using a cable tie around the connector and the socket to lock it in place can help with this.
> > 2. ESD discharge into the EBB resulting in an MCU reset and a disconnect event. To solve this you can have a look at the EBB SB2209 USB manual where there is a section on grounding.
> 
> 1. I use a printed fixing plate. Connection loss does not occur during movement outside the printing process
> 2. I do not use USB, the connection is made via CAN
>    3.The problem only arose recently. Before that, everything had been working smoothly for several months. No changes were made to the configuration.

ESD can affect both USB and CAN.

Also, even a printed strain relief can result in poor connection. Cable ties are the sure way of removing this variable.

ZlobniyDaemon, 2025-04-11:
> ESD can affect both USB and CAN.

Looking [this manual ](https://github.com/bigtreetech/EBB/blob/master/EBB_SB2209_USB/Manual/BIGTREETECH%20EBB%20SB2209%20USB%20V1.0%20User%20Manual.pdf) for grounding information and can't find it. What page i need?

> Also, even a printed strain relief can result in poor connection. Cable ties are the sure way of removing this variable.
In that case, the loss of communication would occur during other movements, such ashoming, input shamer calibration, testing max speed and acceleralions.

ZlobniyDaemon, 2025-04-11:
> https://github.com/bigtreetech/EBB/blob/master/EBB_SB2209_USB/Manual/BIGTREETECH%20EBB%20SB2209%20USB%20V1.2%20User%20Manual.pdf

I followed the grounding instructions, and the connection losses have stopped. I'm still monitoring, but the printing has been running for an hour without any issues.  
Perhaps a similar section on grounding should be added to the SB2206 CAN manual?

closed 2025-04-11 by ZlobniyDaemon

oleksii-suprun, 2025-08-11:
I have the same issue and am looking for help with fixing it. I have BTT Manta M8P 2.0 connected to EBB SB2209 CAN V1.0 (RP2040) with CAN. Randomly, I'm getting an error "Lost communication with MCU 'EBBCan' Once the underlying issue is corrected, use the "FIRMWARE_ RESTART" command to reset the firmware, reload the config, and restart the host software. Printer is shutdown". This could happen after 3 minutes of printing or after 30-40 minutes.

What I've tried/did so far:
1. Upgraded Klipper to the latest version, flashed the latest version of firmware.
2. As was talked about above, I added grounding for the extruder motor.
3. Unfolded the CAN cable and removed it from the case.
4. Tried checked connections, detached/reattached cables.
5. Disassembled and assembled the toolhead to make sure the extrusion is smooth.
6. Verified voltage on the PCU during the print - stable 24.2V, no drops.
8. Bought another EBB SB2209 CAN V1.0 (RP2040) + cable and replaced everything (board and cable one by one).

But nothing from above helped. Random disconnects continue. Another thing I noticed is that it looks like the toolhead continues moving even after the filament has already stopped extruding. The following picture demonstrates this.

[image]

I also noticed that all stops occur during the solid infill printing.

I'm attaching logs and my assembly photo (I use a cable fastening clip "EBB SB2240_2209 CAN/CAD/Printed_Part_for_CAN_Cable_V1.2.STEP" to immobilize the connection, but it is detached in this photo).

[image]

Log files:

- [klippy.log](https://github.com/user-attachments/files/21709366/klippy.log)
- [moonraker.log](https://github.com/user-attachments/files/21709373/moonraker.log)