bigtreetech/EBB #166: SB2209 RP2204 gpio21 and gpio22 always triggered

ZlobniyDaemon, 2025-04-26:
I am trying to connect filament sensors to GPIO21 and GPIO22, but the sensor signal does not change when they are triggered.

`[filament_switch_sensor Aft_Extruder]
pause_on_runout: false
#   When set to True, a PAUSE will execute immediately after a runout
#   is detected. Note that if pause_on_runout is False and the
#   runout_gcode is omitted then runout detection is disabled. Default
#   is True.
runout_gcode: 
    PAUSE
    RESPOND TYPE=error MSG="Обнаружено отсутствие прутка после экструдера"
#   A list of G-Code commands to execute after a filament runout is
#   detected. See docs/Command_Templates.md for G-Code format. If
#   pause_on_runout is set to True this G-Code will run after the
#   PAUSE is complete. The default is not to run any G-Code commands.
insert_gcode: 
    RESPOND TYPE=error MSG="Обнаружена подача прутка"
#   A list of G-Code commands to execute after a filament insert is
#   detected. See docs/Command_Templates.md for G-Code format. The
#   default is not to run any G-Code commands, which disables insert
#   detection.
event_delay: 3
#   The minimum amount of time in seconds to delay between events.
#   Events triggered during this time period will be silently
#   ignored. The default is 3 seconds.
#pause_delay: 0.5
#   The amount of time to delay, in seconds, between the pause command
#   dispatch and execution of the runout_gcode. It may be useful to
#   increase this delay if OctoPrint exhibits strange pause behavior.
#   Default is 0.5 seconds.
switch_pin: EBBCan:gpio21
#   The pin on which the switch is connected. This parameter must be
#   provided.`

I have several SB2209 RP2204 boards, and I am experiencing the same issue with all of them.

cruzer619, 2025-05-19:
please reference #88 for solution.

ZlobniyDaemon, 2025-05-20:
> please reference [#88](https://github.com/bigtreetech/EBB/issues/88) for solution.

Okay, let's say adding a resistor to GPIO21 works (though it's unclear why the pin doesn't work out of the box on the RP2040 but works on STM). What about GPIO22? It also doesn't work—should I solder a second resistor to it as well?

ZlobniyDaemon, 2025-05-20:
> "^EBBCan:gpio22" so like that

Didn't work foк me